ThingPulse/esp8266-weather-station-color #101: No weather data after WU site update

dwhacks, 2019-04-02:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
it Shows the weather

### Actual behavior
Shows the time, a cloud with a ? inside it, and no info.

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.

```c
// add code here
```
### Weather Station Color version
Old single page version, the only one that works with non touch, and I have been using it for 2 years

### Hardware
Run on Commaborable64 board with ESP12

I got a notification the other day that WU was updating their servers/site and ever since then my weather display hasn't been displaying the data. It shows "updating time" etc every 10 minutes as I have it set, and the time is correct but there is no weather data. It does show the sliver of the moon which I believe is correct.

8<------------------------ END BUG REPORT -------------------------------------

G6EJD, 2019-04-02:
I would not classify this as a bug for the very reasons you’ve quoted there are wide-spread WU server issues at the moment, I expect them to continue for many days yet. My system has been uploading weather data to WU for the last 14-years but currently I can’t read my own data or login to my WU account to try to make updates. So this is all a WU and no code errors to report. My own design weather stations using WU also exhibit similar issues to those you report. This is s since IBM purchase md WU - there is nothing more to say about that IMO !

dwhacks, 2019-04-02:
Sounds likely, my weather station is still reporting to WU and I'm able to login and read my data.
This is my station: https://www.wunderground.com/weather/ca/grand-forks/IGRANDFO8

So what you're saying is; wait it out and maybe they will fix it?

G6EJD, 2019-04-02:
Interesting I can see yours but none of my two Davis units, if I look for Bath, UK I get no server response, I think the servers I use in EU are different, for example I can only see a strange mix of Celsius and Fahrenheit.

Did you refresh your API keys?

I think we have to wait for WU to fix all this, although it’s not been working since last Tuesday in the UK.

marcelstoer, 2019-04-03:
Also see https://thingpulse.com/hello-openweathermap-bye-bye-wunderground/

closed 2019-04-03 by marcelstoer