ThingPulse/esp8266-weather-station-color #102: ISR not inIRAM

starfish1107, 2019-05-20:
have problems with my
older version of weatherstation colour kit  with wemos d1 mini  4M.
clean reinstall arduino ide, compiling and flashing (master, Latest commit 2133809 on 24 Feb) according step by step tutorial
the device is looping / restarting , serial monitor shows:

Connecting to WiFi home1/a1b2c3d4e5
.........Connected...Initializing touch screen...
ISR not in IRAM!

any idea how to get it working?

G6EJD, 2019-05-20:
I think you may be using ESP8266 core 2.5.0 and in that build ISR’s now have to be located in IRAM, if so try downgrading your IDE to say 2.4.2 either that or add the IRAM prefix / macro to tell the compiler to place the ISR into IRAM.

marcelstoer, 2019-05-20:
> try downgrading your IDE to say 2.4.2

Not the IDE, but the ESP8266 toolchain i.e. ESP8266 Arduino core as per https://docs.thingpulse.com/guides/wifi-color-display-kit/#install-the-esp8266-toolchain. We will upgrade our code to be compatible with 2.5.x in the following weeks.

starfish1107, 2019-05-20:
youre right  - now it works again  thanks guys.

closed 2019-05-20 by starfish1107

manj9501, 2019-12-24:
There's one more solution, a more correct one.
You can add modify the declaration of Interrupt Service Routine at **line 31** in 
XPT2046_Touchscreen.cpp after installing the XPT2046_Touchscreen library and change it to
`void ICACHE_RAM_ATTR isrPin(void);`
from
`void isrPin(void);`

marcelstoer, 2019-12-24:
That is an issue with the touch library this app depends upon. Please leave a comment at https://github.com/PaulStoffregen/XPT2046_Touchscreen/pull/24#issuecomment-531621444 asking the maintainer to please release the fix he committed to master back in June.

MagnusThome, 2020-03-10:
Any updates on this other than downgrading ESP8266 core to before 2.5.0?

MagnusThome, 2020-03-10:
Manually updated the https://github.com/PaulStoffregen/XPT2046_Touchscreen/ since it isn't updated from the Arduino IDE library manager. Seem to work fine now :-)

marcelstoer, 2020-03-10:
See my previous comment. If enough people ask for a new release the maintainer may eventually create one.