ThingPulse/esp8266-weather-station-color #103: Replacing the Touch Function with Push Button

Dancopy, 2019-06-05:
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

### Actual behavior

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.

```c
// add code here
```
### Weather Station Color version
Which branch are you on? If you know the Git revision then add it here as well.

### Hardware
Describe whether you run the code on the ThingPulse Weather Station Color kit or on some custom hardware.

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

Dancopy, 2019-06-05:
Does anyone know how to replace the Touch function with Push Button with resistor PULL_UP in code, maybe on Pin D3?
Cause: My screen (ili9341 2.8") does not have a Touch function.
Thank you

plazmax, 2019-07-05:
im curious too

mazur888, 2019-07-28:
Thinking about the same. I think it would be better than touch screen.

closed 2020-02-08 by stale[bot]