ThingPulse/esp8266-weather-station-color #105: Connecting to 3.2" TFT without serial interface

DrKuebel, 2019-08-03:
8<------------------------ FEATURE REQUEST ------------------------------------
### Connecting to 3.2" TFT with ILI9341 an 16 /8 Bit is not possible 

### Adding additional Wiringlayout for bigger Display without serial connector

https://de.aliexpress.com/item/32808968215.html?spm=a2g0o.productlist.0.0.398c4cbbl4g1Fy&algo_pvid=4994521d-4bff-4bbd-a48f-a018376616d3&algo_expid=4994521d-4bff-4bbd-a48f-a018376616d3-7&btsid=c5e76fb1-ca88-451d-bb1f-8f54027e1197&ws_ab_test=searchweb0_0,searchweb201602_7,searchweb201603_52

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

G6EJD, 2019-08-03:
Why not buy the SPI (serial) version it’s half the price and speed nearly the same?

DrKuebel, 2019-08-05:
I accidentialy bought the wrong one ;-)

G6EJD, 2019-08-05:
Your problem (parallel interface) is there are not enough IO pi s to drive the display even on a (say) NodeMCU. You’d need 8-data plus a clock and chip-select and a command/data selector, about 11, maybe more with reset and busy detection, that’s a total of 13, this is SPI is more GPIO efficient. You could use the display on a UNO/MEGA if pin compatible.

closed 2020-02-15 by stale[bot]