ThingPulse/esp8266-weather-station-color #108: Ordered new screen, wired up and got it turning on. Now the system resets after WiFi Connected.

bizhostnc, 2019-09-28:
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
System starts to display weather data, etc. First setup.
### Actual behavior
Shows me WiFi Connected, then the blue LED on the ESP8266 lights up and the system restarts connecting to WiFi.
### Test code
Stock code from latest download.

bizhostnc, 2019-09-28:
If I comment out this part of the code, it stops the AM/PM and time blinking. But then it adds a small time reading in smaller font next to the main time clock, closer to the WiFi signal. I will continue to look at this. Not sure why I am the only one reporting it; the clock seems to work fine it's just the AM/PM part causing a blink. Wondering if it's the IS_STYLE_12HR setup.

 // if (IS_STYLE_12HR) {
 //   sprintf(time_str, "%s\n%s", dstAbbrev, timeinfo->tm_hour>=12?"PM":"AM");
 // } else {
 //   sprintf(time_str, "%s", dstAbbrev);
 // }

bizhostnc, 2019-09-28:
Definitely something with the (IS_STYLE_12HR) going on, if I comment out the options, and force it to display one way or the other. It does not blink and shows fine.

 // if (IS_STYLE_12HR) {
    int hour = (timeinfo->tm_hour+11)%12+1;  // take care of noon and midnight
    sprintf(time_str, "%2d:%02d:%02d\n",hour, timeinfo->tm_min, timeinfo->tm_sec);
 // } else {
  //  sprintf(time_str, "%02d:%02d:%02d\n",timeinfo->tm_hour, timeinfo->tm_min, timeinfo->tm_sec);
 // }
  gfx.drawString(120, 20, time_str);

  gfx.setTextAlignment(TEXT_ALIGN_LEFT);
  gfx.setFont(ArialMT_Plain_10);
  gfx.setColor(MINI_BLUE);
 // if (IS_STYLE_12HR) {
    sprintf(time_str, "%s\n%s", dstAbbrev, timeinfo->tm_hour>=12?"PM":"AM");
   // Serial.printf("It's printing 12hr style");
  //} else {
   // Serial.printf("It's printing 24hr style");
 //   sprintf(time_str, "%s", dstAbbrev);
 // }

plazmax, 2019-09-28:
Interesting i ll try it, ty

bizhostnc, 2019-10-30:
It's been working since I commented out the option to change from 12 to 24hr, it broke the AM/PM for some time until I moved a few things

jingsno, 2019-11-02:
Changing both _`if (IS_STYLE_12HR)`_ to `if (false)` also fixes the problem

Anon0ne, 2019-11-02:
> Changing both _`if (IS_STYLE_12HR)`_ to `if (false)` also fixes the problem

Not when I done it. It caused the blinking AM or PM depending on the time.

G6EJD, 2019-11-02:
Have you tried sending the value of IS_STYLE_12HR to the serial monitor Just before the test, to see if it’s value has been modified eg serial.println(IS_STYLE_12HR);
This could be a compiler issue, what version of the ESP core are you using?

Anon0ne, 2019-11-02:
Whatever version was recommended here on forum and in code.

Justin@BizHostNC.com
BizHostNC.com - Professional Business Web Hosting & Internet Marketing

Technical & PC Remote Support
https://bizhostnc.com/clients
704-699-9652 (7AM-7PM M-F)

pprice-meow, 2020-03-25:
The problem I had was that the hardware for the touchscreen is not supported by the latest drivers. I  had to go back to 1.1.0 and remove the IRQ from the touchscreen object. When I started the screen with ts.begin(), my esp8266 crashed. If I left that out, it would run fine. The screen would not react to a touch on the screen. The time would switch in and out of 12/24hr format.

G6EJD, 2020-03-25:
Did you try moving the IRQ to ISRAM with the compiler directive

c-klinger, 2020-04-09:
@G6EJD can you explain your idea more in detail?

I had the same issue (using an non-touch display) and could also fix it by changing the line:
`XPT2046_Touchscreen ts(TOUCH_CS, TOUCH_IRQ);`
into
`XPT2046_Touchscreen ts(TOUCH_CS);`
as suggested by @pprice-meow. 

Looking into XPT2046_Touchscreen source i realized the same can be archived by just changing the TOUCH_IRQ to 255 in settings.h which seems to be the 'cleaner' way of doing it.

G6EJD, 2020-04-09:
Well my thinking is there’s no doubt that interrupt routines should not interfere with the main routine (before IRQ took place) and for a long time the compiler never separated the memory, but now it does, so adding ICACHE_RAM_ATTR directive before the IRQ function name, eg void directive function_name Okay SD the IRQ handler in memory where it will not cause any conflicts.
Also to note that there are other IRQ’s running in normal use, so I can see how your solution of moving the index up and probably up the IRQ stack could solve the problem, but it’s not necessarily the right solution.

I thought I’d seen that ThingPulse had implemented the IRAM fix in a recent release and your compiler if the latest version should warn you of the error.

Scissorunner, 2020-04-15:
I too am seeing the time blink back and forth from 12hr to 24hr format. The ILI9341 TFT LCD display I'm using does not include the chip used for the touch screen.

I read through the previous comments and tried the suggestions from @pprice-meow and @c-klinger and was not successful in stopping the blink pattern.

Believe my best recourse is to purchase another display that includes the touch screen, even though for this project I do not understand what functionality this would bring.

If anyone has further suggestions I would be grateful for guidance. Thank you

Anon0ne, 2020-04-15:
You can force 12hr. It's likely not the display but the IDE.

Justin@BizHostNC.com
BizHostNC.com - Professional Business Web Hosting & Internet Marketing

Technical & PC Remote Support
https://bizhostnc.com/clients
704-699-9652 (7AM-7PM M-F)

Scissorunner, 2020-04-15:
@Anon0ne Wow! thank you so much, just needed a little brain push to think in a different direction.

Starting on line 230 in main code I changed IS_STYLE_12HR = !IS_STYLE_12HR; to IS_STYLE_12HR = IS_STYLE_12HR; (removed the !)

That's all that was needed. No more blinking time!!!

if (p.y < 80) {
      IS_STYLE_12HR = IS_STYLE_12HR; // 2nd was !IS_STYLE_12HR
    } else {
      screen = (screen + 1) % screenCount;

Thanks agan!!

Anon0ne, 2020-04-16:
No problem. I battled same problems

Justin@BizHostNC.com
BizHostNC.com - Professional Business Web Hosting & Internet Marketing

Technical & PC Remote Support
https://bizhostnc.com/clients
704-699-9652 (7AM-7PM M-F)

mickey-zg, 2020-07-22:
I had the same problem. I found that my touch screen could not be initialized
My Configuration:
ESP8266 (Community Version 2.7.1)
ILI9341SPI
Board: LOLIN(WEMOS ) D1 R2 & mini

Message on Serial Monitor:
................Connected...Initializing touch screen...
ISR not in IRAM!

Abort called

Because of this, the sketch constantly thinks the touchscreen has been touched, and constantly changes the AM / PM time display.
The solution is in the \ your_arduino_library_folder \ XPT2046_Touchscreen \ XPT2046_Touchscreen.cpp file.
Line 49 needs to be changed from
**_void isrPin( void )_**
to 
_**ICACHE_RAM_ATTR void isrPin( void )**_

the whole section looks like this

_**#ifdef ESP32
void IRAM_ATTR isrPin( void )
#else
//void isrPin( void )
ICACHE_RAM_ATTR void isrPin( void )
#endif**_

After I did that, the touchscreen worked properly, and with it also the code.

marcelstoer, 2020-07-22:
The "ISR not in IRAM!" is unrelated to the OP's issue. Time to close it anyway.

> The solution is in the \ your_arduino_library_folder \ XPT2046_Touchscreen \ XPT2046_Touchscreen.cpp file.

Please don't do this. It's not the recommended fix. The proper solution is to read the documentation linked in the README and use "ThingPulse XPT2046 Touch" (forked from Paul Stoffregen).

closed 2020-07-22 by marcelstoer

yemu, 2021-02-24:
> @Anon0ne Wow! thank you so much, just needed a little brain push to think in a different direction.
>
> Starting on line 230 in main code I changed IS_STYLE_12HR = !IS_STYLE_12HR; to IS_STYLE_12HR = [...]

for me that solved the blinking issue, but the touch is not working unfortunately

artemijkulakov, 2021-09-30:
It helped me: 
​if (p.y != 0) {
IS_STYLE_12HR = IS_STYLE_12HR;
} else {
screen = (screen + 1) % screenCount;
}