ThingPulse/esp8266-weather-station-color #109: MoonAge wrong

bernemer, 2019-10-11:
Hi,
at my installation i habe a Worng MoonAge.
Today at Oct 11 2019 12:35 CEST the Station say we have Fullmoon (Age 14d, Illum 97%) but Fullmoon is at Oct 13 2019 at 23:07CEST.

Time an Date are ok.

Anybody an Idea?

G6EJD, 2019-10-11:
It’s an approximation

bernemer, 2019-10-11:
You have the same effect?
2 1/2 days is a lot.

bizhostnc, 2019-10-30:
Believe this is affected by the channel you choose in weather API. You must choose one that is most suitable to your actual conditions, hence this is why I have a local station, and a WiFi station. Eventually I may try to log the current information to internet, but not now.

G6EJD, 2019-10-30:
Moon age is usually derived as it is not generally sent in the API, it is possible the calculations are slightly in error.

bernemer, 2019-12-19:
If i understand the code right, the moonage is bulit from the calculation of illumination .
But the illumination of the Moon is not linear. 
So in the near of full and new moon the moonage is high.

Thanks for the answers

I close the issue with no correction.

closed 2019-12-19 by bernemer