ThingPulse/esp8266-weather-station-color #110: Add "Feels-Like" Temperature

sungkhum, 2019-11-18:
8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature
There is currently no "feels-like" temperature displayed.
### Justification
Personally, that is a number I like to see. Could be nice for others as well.

### Workarounds
Here's the code the calculate it. I put it in the void drawCurrentWeather() function and replaced the city name text with what the temperature feels like. Converted the code from here: https://gist.github.com/jfcarr/e68593c92c878257550d

`  //Calculate "Feels-Like" Temp
  float vTemperature = currentWeather.temp;
  float vWindSpeed = currentWeather.windSpeed;
  float vRelativeHumidity = currentWeather.humidity;
  float vFeelsLike = 0.0;
  //Try Wind Chill First
  if (vTemperature <= 50 && vWindSpeed >= 3) {
    vFeelsLike = 35.74 + (0.6215*vTemperature) - 35.75*(pow(vWindSpeed,0.16)) + ((0.4275*vTemperature)*(pow(vWindSpeed,0.16)));
  } else {
    vFeelsLike = vTemperature;
  }
  //Use Heat Index if Needed
  if (vFeelsLike == vTemperature && vTemperature >= 80) {
    vFeelsLike = 0.5 * (vTemperature + 61.0 + ((vTemperature-68.0)*1.2) + (vRelativeHumidity*0.094));
    if (vRelativeHumidity < 13 && vTemperature >= 80 && vTemperature <= 112) {
      vFeelsLike = vFeelsLike - ((13-vRelativeHumidity)/4)*(sqrt((17-(fabs(vTemperature-95.)))/17));
      if (vRelativeHumidity > 85 && vTemperature >= 80 && vTemperature <= 87) {
        vFeelsLike = vFeelsLike + ((vRelativeHumidity-85)/10) * ((87-vTemperature)/5);
        
      }
      
    }
  }`

8<------------------------ END FEATURE REQUEST --------------------------------

marcelstoer, 2019-11-18:
That would be a useful piece of information, I agree. However, it would be a lot more useful if it were built into the https://github.com/ThingPulse/esp8266-weather-station library that this project also relies upon. Would you mind raising a PR there?

closed 2019-11-18 by marcelstoer

sungkhum, 2019-11-19:
@marcelstoer Thanks for getting back to me. That makes sense. I raised a PR for the main library. I don't know C++ so I don't know if I put it in the best place, but it works for me.

G6EJD, 2019-11-19:
The problem with your code example is that it only works for °F and therefore it needs modifying to cope with either °C or °F can the units variable

sungkhum, 2019-11-19:
Ah, yes, you are right - thanks @G6EJD I'll modify the code.

sungkhum, 2019-11-19:
I modified the code and made a new pull request https://github.com/ThingPulse/esp8266-weather-station/pull/173

@G6EJD  @marcelstoer