ThingPulse/esp8266-weather-station-color #112: Issues with moving past connecting phase

John8989doe, 2019-11-24:
I just got this working but now it only shows the starting screen where it says connecting to my WiFi, then it will say Connected. After that the screen goes blank and comes back up with the start screen saying connecting to my WiFi

G6EJD, 2019-11-24:
Do you have a valid api key?

John8989doe, 2019-11-25:
Yes, I have a valid api key

manoranjan2050, 2019-12-03:
please check WiFi connection

marcelstoer, 2019-12-03:
Without the output from your serial console it's unlikely we'll be able to help you.

John8989doe, 2019-12-05:
Serial Monitor:

Connecting to WiFi "SSID"/"Password"
......Connected...Initializing touch screen...
ISR not in IRAM!

Abort called

>>>stack>>>

ctx: cont
sp: 3ffffd90 end: 3fffffc0 offset: 01b0
3fffff40:  009c1001 3ffef778 3ffef778 4020a569  
3fffff50:  3ffe9001 fffef778 00000000 3ffef894  
3fffff60:  3fffdad0 000000ff 3ffe8534 401004c6  
3fffff70:  014000f0 3ffef778 3ffe8534 40205320  
3fffff80:  3fffdad0 3ffef778 3ffef4f0 40201cdc  
3fffff90:  feefeffe feefeffe feefeffe feefeffe  
3fffffa0:  3fffdad0 00000000 3ffef864 4020bfac  
3fffffb0:  feefeffe feefeffe 3ffe859c 401007b1  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
15

I checked, rechecked and checked again the settings.h to make sure my WiFi was correct and it is.

marcelstoer, 2019-12-05:
> ISR not in IRAM!

As per https://docs.thingpulse.com/guides/wifi-color-display-kit/#install-the-esp8266-toolchain you should be using ESP8266 Arduino core **2.4.2**. It appears that you're on a newer version.

The issue is caused by the XPT2046 touch library which doesn't support newer versions yet. Please leave a comment with https://github.com/PaulStoffregen/XPT2046_Touchscreen/pull/24 to motivate the author to finally release the fix that he committed months ago.

closed 2019-12-05 by marcelstoer

erickbyrne, 2020-04-15:
Hello Marcelstoer,  I have a similar error with my  esp8266-weather-station-color.
but I have the following error.

ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
15
Connecting to WiFi TP-Link_2CD1/xxxxxxxxx
......Connected...Initializing touch screen...
Mounting file system...
SPIFFS opened: OK
.
Current time: 1586914953

DST Rules Updated:
DST Start: Sun Mar 29 02:00:00 2020
DST End:   Sun Oct 25 02:00:00 2020

Time difference for DST: 7200
[HTTP] Requesting resource at http://api.openweathermap.org:80/data/2.5/weather?id=3530597&appid=e607fead7e3d643fe06b080b84bce4c9&units=metric&lang=en
[HTTP] connected, now GETting data
start document
[HTTP] Requesting resource at http://api.openweathermap.org:80/data/2.5/forecast?id=3530597&appid=e607fead7e3d643fe06b080b84bce4c9&units=metric&lang=en
[HTTP] connected, now GETting data
start document
Free mem: 19880