ThingPulse/esp8266-weather-station-color #113: UTC_OFFSET errror

manoranjan2050, 2019-12-03:
Hello Team And Friend 

i'm from india My UTS Is IST +5.30 
which upset i use when i change like this 
UTC_OFFSET +5.30 its show incorrect time please help me i also try with india ntp pool server but problem same please help me

G6EJD, 2019-12-03:
Have you tried 5.5 rather than 5.30 it’s a decimal value so 30/60 = 0.5

manoranjan2050, 2019-12-04:
Many Thanks For Great Help i tested With Your Method , 
now problem is my local time is 8:03 Am but Device Show 30 Min late 7:33 Am Please Guide Me

G6EJD, 2019-12-04:
Yo need to increase or decrease the gmt offset until your time zone is correct, as 5.5 is displaying 07:30 instead of 08:00 then try 6.0 I suspect the time server your using is wrong for your region.

manoranjan2050, 2019-12-04:
May Thanks For Faster response sir ... yes i tested with 6.0 but when i use 6.0 its show 30 min fast time if my local time is 10.30 Am Its show me 11 am and i use 
0.in.pool.ntp.org server ,

nestlerw, 2020-02-26:
India does not follow DST and we are using the library simpleDSTadjust. I put this in settings.h and it appears to work. 

// TZ Settings for India
#define UTC_OFFSET 5.5 
struct dstRule StartRule = {"IST", Second, Sun, Mar, 2, 0}; // India Standard Time = UTC/GMT +5.5 hours
struct dstRule EndRule = {"IST", First, Sun, Nov, 1, 0};    // India Standard time = UTC/GMT +5.5 hours

closed 2020-03-15 by marcelstoer