ThingPulse/esp8266-weather-station-color #115: Only White screen

Desterly, 2020-01-10:
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Screen inits and functions
### Actual behavior
No screen init - only backlight
### Test code
Latest versions of esp8266-weather-station-color and dependencies.  
```c
// add code here
```
### Weather Station Color version
Which branch are you on? If you know the Git revision then add it here as well.
master
### Hardware
Color weather station kit received from ThingPulse (LOLIN(WEMOS) D1 R2 & mini)

With the configuration and upload following the guide on https://docs.thingpulse.com/guides/wifi-color-display-kit/ the sketch uploads correctly.

Serial output shows it's connecting to Wifi and getting data from openweather however the display itself is only white which leads me to believe that only the backlight is turning on.

I know there has been issues w/ the later versions of arduino core which required some modifications to the touch lib.   Is there a recommended core version or should the latest work?  Is it possible this is a bad display?

8<------------------------ END BUG REPORT -------------------------------------

closed 2020-02-23 by Desterly

TBR-BRD, 2020-09-30:
I have the same Problem, please help. Thank you

ESP32

cheng3100, 2021-01-27:
Hi, I don't know why this issue is closed but I meet same problem which I solved it finally. It is very stupid. Just because the REET pin of ILI9341 not be connected. I just add line like this:
```
// modify code in setting.h
// Pins for the ILI9341
#define TFT_DC D2
#define TFT_CS D9
#define TFT_LED D3
#define TFT_RESET D8  // add a reset pin

#define HAVE_TOUCHPAD
#define TOUCH_CS D10
#define TOUCH_IRQ  D4

```
and modify one in ino file to register the reset pin :
```
ILI9341_SPI tft = ILI9341_SPI(TFT_CS, TFT_DC, TFT_RESET); 
```

I don't know why everyone else didn't  mentioned this . I check the code of SPI.cpp for esp8266. The rest pin only work once before spi initialization :
[image]

It shouldn't work if the reset PIN is not toggled like this .

marcelstoer, 2021-01-28:
@cheng3100 Thanks for your input. How does that match with the ThingPulse connector board spec https://docs.thingpulse.com/specs/wifi-color-display-kit/ that has `D8 -> TFT LED`?

cheng3100, 2021-01-28:
@marcelstoer  Hi, I use a wemos D1 board which like this one:
[image]. 
I change the pinout of in settting.h because I found that the the `D1` pin is also the uart `TX` pin in wemos D1, which will make the uart output error. So I just modify the pinout and add the RESET pin.

cheng3100, 2021-01-28:
@marcelstoer  And I check the `pins_arduino.h` file inside the arduino board lib. I found that the `wemos` 's pin map is very strange and different from others, wemos pin map is here:
[image]
You can see that it map the `D1` and `TX` with the same gpio pin 1.

But most others pin map like this:
[image]
So I think maybe we add a macro iswitch n the `setting.h` for different board and different pin map.

marcelstoer, 2021-01-28:
> add a macro switch in the `setting.h` for different board and different pin map

The primary intention for this code is work as-is with our IoT kit [as described in the README](https://github.com/ThingPulse/esp8266-weather-station-color/blob/master/README.md). The community is of course invited to add features that make the use with other hardware more convenient.