ThingPulse/esp8266-weather-station-color #116: Forecasts not updated properly on cold boot and reset.

manj9501, 2020-02-29:
Whenever I cold boot my setup or reset it, only a few number of forecasts are updated, with the rest appearing blank as shown below in attached pictures

[image]
[image]

After one or two updates of the forecasts after the specified UPDATE_INTERVAL_SECS, the issue is usually resolved.
Can someone please point me to what I am doing wrong?

ackbad, 2020-03-10:
I just built my own yesterday and have the exact same issue.

marcelstoer, 2020-03-10:
> Can someone please point me to what I am doing wrong?

Maybe it's not you but the ESP8266 Arduino Core. Which version are you using? https://docs.thingpulse.com/guides/wifi-color-display-kit/#install-the-esp8266-toolchain

manj9501, 2020-03-10:
I am using version, **2.6.3**, of the ESP8266 Arduino Core, which happens to be the latest, as of now.
And yes, I am using the version **2.0.0** of the ESP8266 Weather Station library, as I could see from the Arduino Library Manager.

Plus, I want to add that that I tested the WeatherStationDemo example of the ESP8266 Weather Station library with my 0.96" I2C OLED display and observed similar anomalous behavior, wherein forecasts were not updating properly.

marcelstoer, 2020-03-10:
Thanks for that extra information. I guess the behavior is not surprising considering that the parsing & mapping of the data happens in the weather station library. -> I'll transfer the issue