ThingPulse/esp8266-weather-station-color #118: User exception (ISR is nor IRAM) rst cause:2, boot mode:(3,6)

karmacoma92, 2020-03-19:
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
continue working
### Actual behavior

20:25:28.411 -> ..........Connected...Initializing touch screen...
20:25:34.183 -> ISR not in IRAM!
20:25:34.183 -> 
20:25:34.183 -> User exception (panic/abort/assert)
20:25:34.183 -> Abort called
20:25:34.183 -> 
20:25:34.183 -> >>>stack>>>
20:25:34.183 -> 
20:25:34.183 -> ctx: cont
20:25:34.183 -> sp: 3ffffed0 end: 3fffffc0 offset: 0000
20:25:34.183 -> 3ffffed0:  3ffe89f7 00000000 002e2e2e 4020ecc5  
20:25:34.216 -> 3ffffee0:  000000fe 00000000 00000000 00000000  
20:25:34.216 -> 3ffffef0:  00000000 00000000 00000000 00ff0000  
20:25:34.216 -> 3fffff00:  5ffffe00 5ffffe00 3ffe89eb 00000000  
20:25:34.216 -> 3fffff10:  00000002 00000002 3ffe8528 4020d2d2  
20:25:34.216 -> 3fffff20:  401004d6 00000000 4d002e2e 4020d2e8  
20:25:34.216 -> 3fffff30:  3ffef9d8 00000000 3ffe8528 4020d7ed  
20:25:34.216 -> 3fffff40:  00000000 3ffefa24 3ffe9096 4020af4c  
20:25:34.250 -> 3fffff50:  4020af40 3ffefa24 3ffe9096 3ffefb54  
20:25:34.250 -> 3fffff60:  3fffdad0 000000ff 3ffe8528 4020d89c  
20:25:34.250 -> 3fffff70:  3fffdad0 3ffefa24 3ffe8528 40205b29  
20:25:34.250 -> 3fffff80:  3fffdad0 3ffefa24 3ffef770 40201e69  
20:25:34.250 -> 3fffff90:  feefeffe feefeffe feefeffe feefeffe  
20:25:34.250 -> 3fffffa0:  3fffdad0 00000000 3ffefb14 4020ccdc  
20:25:34.250 -> 3fffffb0:  feefeffe feefeffe 3ffe8580 40100e31  
20:25:34.283 -> <<<stack<<<
20:25:34.283 -> 
20:25:34.283 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
20:25:34.283 -> 
20:25:34.283 -> load 0x4010f000, len 1392, room 16 
20:25:34.283 -> tail 0
20:25:34.283 -> chksum 0xd0
20:25:34.283 -> csum 0xd0
20:25:34.283 -> v3d128e5c
20:25:34.283 -> ~ld
20:25:34.349 -> 15

### Test code
add :

```
// add code here
  ts.begin();
```
### Weather Station Color version
Master branch

### Hardware
I'm running it on custom HW

8<------------------------ END BUG REPORT -------------------------------------

marcelstoer, 2020-03-19:
https://docs.thingpulse.com/guides/wifi-color-display-kit/#install-the-esp8266-toolchain

closed 2020-03-19 by marcelstoer

karmacoma92, 2020-03-19:
thanx!!!!!