ThingPulse/esp8266-weather-station-color #119: Can't connected wifi

page4me, 2020-05-25:
I can't connect to wifi The settings are correct: ssid and password api key. All time is connected to WIFI "myssid" preloader goes to the end and shows Connected to WIFI "myssid" and so nothing else is displayed all the time. Wifi data entered in the settings and esp8266 file manula connect wifi and the connection does not work.

WOnken, 2020-06-04:
I am having exact the same issue... And this happened after an update. Not a brand new installation. Before that it was running. My Arduino IDE as well as all libraries are absolutely up to date. There must be an error in the code

G6EJD, 2020-06-04:
Earlier versions of the IDE allowed an interrupt service routine to not reside in IS RAM, so this is why the issue is occurring, if your downgrade to a an IDE version maybe 2.4.2 it will be OK until the source code is modified to include the IRAM prefixes, actually I thought they had been updated, are you sure you have the latest source code?

marcelstoer, 2020-06-04:
The issue title is "Can't connected wifi". Yet the console log apparently says

```
Connecting to WiFi P4 2G/44FCWXRC
Connected...
```
So, no, WiFi connection is not an issue.

> actually I thought they had been updated

Yes, they have. If folks bothered to read the documentation linked in the README, particularly https://docs.thingpulse.com/guides/wifi-color-display-kit/#install-libraries, they wouldn't have this issue.

closed 2020-06-04 by marcelstoer