ThingPulse/esp8266-weather-station-color #122: how to use changeBitDepth

Aleks-Ale, 2020-12-13:
Dear Sirs,
could you please explain how to use changeBitDepth? 
I want to use more than four colors in the esp8266-weather-station-color project. 
-.
uint16_t palette [] = {ILI9341_BLACK, // 0
                      ILI9341_WHITE, // 1
                      ILI9341_NAVY, // 2
...
                      ILI9341_MAGENTA, // 14
                      0xFD80}; // 15
int BITS_PER_PIXEL = 4;
-.
But the current weather icons (weathericons.h) have BitDepth: 2.
I've tried using changeBitDepth:
-.
void drawCurrentWeather () {
  uint16_t palette [] = {ILI9341_BLACK, // 0
                      ILI9341_WHITE, // 1
                      ILI9341_YELLOW, // 2
                      0x7E3C
                     }; // 3
  gfx.changeBitDepth (2, palette);
  gfx.drawPalettedBitmapFromPgm (0, 105, getMeteoconIconFromProgmem (currentWeather.icon));
-.
But it doesn't work (Bmp has wrong bit depth). 
How can I solve this problem?
Thank you.

G6EJD, 2020-12-13:
Weather icons are fixed bit-depth you have to create new icons with a wider colour range. The bit depth your referencing is for other functions like background colours.

Aleks-Ale, 2020-12-13:
Thank you so much for quick answer!
Could you please advise the software for convert bmp to const char like weathericons.h in this project?

G6EJD, 2020-12-13:
https://sourceforge.net/projects/lcd-image-converter/files/

Aleks-Ale, 2020-12-13:
Thank you very much! 
I will try to use this converter.

Aleks-Ale, 2020-12-13:
I have tried different settings of the LCD Image Converter and LCD Assistant but could not get a good result. 
Should this really work for TFT ILI9341? 
For test, I tried to get the same code when converting "tstorms.bmp" with a depth of 2 bits per pixel. 
But none of the attempts resulted in the correct image on the screen. 
Some dots and stripes are displayed, but this is not the original image.
Does anyone have a successful experience of using any converter for bmp for TFT? 
I will be very grateful for any help.

G6EJD, 2020-12-13:
creating images for display is not easy. You have to select the correct direction and format I think this software uses a special image format, you need to check what driver is being used. XBM I recall

Aleks-Ale, 2020-12-13:
Thank you for your attention to my question. I am trying to get an image for esp8266-weather-station-color which uses Mini Grafx by Daniel Eichhorn. The sketch archive has a folder of used bmp images.
[image]
Also the file 'weathericons.h' contains the code for these images.
const char tstorms[] PROGMEM = {
  0x01, // Version: 1
  0x02, // BitDepth: 2
  0x00, 0x62, // Width: 98
  0x00, 0x64, // Height: 100
  // Round width to next byte: 104
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
I want to find a converter for such a conversion.

marcelstoer, 2020-12-13:
> I want to find a converter for such a conversion.

Try this https://github.com/squix78/bitmapconverter-for-minigrafx

Aleks-Ale, 2020-12-13:
This is definitely another great work by Daniel Eichhorn! 
But only one problem - how to use it with Windows? :)
I tried to use command line to run java
java -cp ./ Converter.java sleet.bmp > sleet.c
Error.
This should probably have a different way to run or additional infrastrucrure.

lorol, 2020-12-14:
Hi, See also this.
https://github.com/ThingPulse/esp8266-weather-station-color/issues/41#issuecomment-388780985

Aleks-Ale, 2020-12-14:
> Hi, See also this.
> [#41 (comment)](https://github.com/ThingPulse/esp8266-weather-station-color/issues/41#issuecomment-388780985)

Thank you so much! I will try. 2 bit depth only? This is possible for 4 bit depth ?
Any way, thank you again!

Aleks-Ale, 2020-12-15:
> Hi, See also this.
> [#41 (comment)](https://github.com/ThingPulse/esp8266-weather-station-color/issues/41#issuecomment-388780985)
Hi, lorol, how can I contact you with another question?

lorol, 2020-12-15:
@Aleks-Ale  You can post it here, to #41 or create a new case whatever is more appropriate

Aleks-Ale, 2020-12-15:
> @Aleks-Ale You can post it here, to #41 or create a new case whatever is more appropriate

I need your help here:
https://github.com/lorol/esp8266-oled-ssd1306/tree/master/resources/FontConverterV3
russian?

Aleks-Ale, 2020-12-15:
> @Aleks-Ale You can post it here, to #41 or create a new case whatever is more appropriate
Is it possible to display the Cyrillic fonts in Mini Grafx by Daniel Eichhorn for TFT ili 9341 (esp8266-weather-station-color)?

lorol, 2020-12-15:
I don't remember if I did try Cyrillic fonts adapting on color project. 
You need to replace ArialRounded.h - yes you can generate yourself by https://github.com/lorol/esp8266-oled-ssd1306/tree/master/resources/FontConverterV3
or use https://forum.arduino.cc/index.php?topic=447956
Squix V3 format I believe.
Then use UTF-8 file encoding for your sketch and write the strings as you wish.
It may be an additional trick for 2 symbols but I don't remember anymore.

Aleks-Ale, 2020-12-15:
I have two unresolved problems for the weather station. 
1. Using Cyrillic to display weather descriptions (ru) from OpenWeatherMap. 
2. Using more than 4 colors for screen decoration. Weather icons can be 2-bit depth.

> Then use UTF-8 file encoding for your sketch and write the strings as you wish.
I tried to use the Arial_c_10.V3.h Cyrillic font and the MiniGrafx utf8ascii function. Nothing. 
I do not know what to do.
Any ideas?

commar, 2021-01-29:
1. Please help.
My station is flashing time (12/24h), why is that?
https://www.youtube.com/watch?v=Yk5BN8K-zUs

2.
How do I edit the sketch for the ILI9488 display (320x480)?
Thank Martin

closed 2021-08-21 by stale[bot]