ThingPulse/esp8266-weather-station-color #124: 12 hour setting on clock but not on forecasts

SillyHats, 2021-03-30:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Clock on top shows AM/PM with tk setting set to tk. Currently the clock shows 6:15pm. Under clock is current conditions. Under current conditions is the forecast scrolling right to left. Under the forecast is the moon information with rise and set times. 

### Actual behavior
The times for the forecast currently show 8:00 and 20:00 rather than 8am and 8pm. Moon rise and set times show 06:00 and 19:00 rather than 6:00am and 7:00pm.

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.

```c
// add code here
```
### Weather Station Color version
Master code branch. 

### Hardware
2.8" TFT touch display with ESP8266 D1 pro

8<------------------------ END BUG REPORT -------------------------------------

TiVoHomeUser, 2021-04-18:
I created a fix for this and made a draft pull request "Time period" with proposed changes please check it out

SillyHats, 2021-04-27:
[PR#128](https://github.com/ThingPulse/esp8266-weather-station-color/pull/128) seems to have fixed this issue. Thanks!

While we are here discussing units do you know which line controls the decimal precision of the temperature display in current and forecasts? I would prefer 65F to the 65.1F that it currently shows.

marcelstoer, 2021-04-27:
> do you know which line controls the decimal precision of the temperature display in current and forecasts?

https://github.com/ThingPulse/esp8266-weather-station-color/blob/master/esp8266-weather-station-color.ino -> search for "temp"

PR #128 (2021-12-25): Time period
Fixes #124 & #127.

- [x] This PR is compliant with the [other contributing guidelines](https://github.com/ThingPulse/esp8266-weather-station-color/blob/master/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] Should this code require changes to documentation I will contribute those to [https://github.com/ThingPulse/docs](https://github.com/ThingPulse/docs).

 - Fix for broken sleep mode
 - Soft sleep mode
 - Fix for 12/24 hr displays not changing
commit 5b34146e (2021-12-25): Add fixes for deep sleep and 12/24h time formatting (#128)

- There was no way to stop the initiated sleep process after sleep timeout screen started
- Introduce soft sleep mode where just the backlight is turned off
- Fix inconsistent 12/24h time formatting  on some screens

Fixes #124 & #127.
closed 2021-12-25 by marcelstoer