ThingPulse/esp8266-weather-station-color #125: TouchControllerWS::isTouched() missing return

TiVoHomeUser, 2021-04-14:
Return value missing from function in TouchControllerWS.cpp causing random results 

line 90  TouchControllerWS.cpp master branch as of April 14 2021
```c
bool TouchControllerWS::isTouched() {
  touchScreen->touched();
}
```

Proposed change:
```c
bool TouchControllerWS::isTouched() {
 return touchScreen->touched();
}
```
### Expected behavior

function TouchControllerWS::isTouched(void) return true or false depending on screen touch.

### Actual behavior

The return value from function bool TouchControllerWS::isTouched() is undefined not always agrees with touchscreen actions

### Test code
I attempting to check for a screen touch to interrupt going into the "Now in sleep mode" using the isTouched(void) function I was getting unreliable results. Tested by putting in repeating loop
```c
if( true == touchController.isTouched() ) Serial.println("True"); else Serial.println("False");
```
### Weather Station Color version
download from ThingPulse  master branch April 14 2021

### Hardware
ThingPulse Weather Station Color kit purchased through Amazon March 2021

G6EJD, 2021-04-14:
Interesting, it was OK with the earlier ESP8266 compiler and I’m surprised the current version isn’t reporting an error, the ESP32 variant would.

marcelstoer, 2021-04-15:
It's frustrating a compiler would ever accept such code, sigh. @TiVoHomeUser willing to create a PR?

TiVoHomeUser, 2021-04-15:
> It's frustrating a compiler would ever accept such code, sigh. @TiVoHomeUser willing to create a PR?

Done (I think)

PR #126 (2021-04-15): Update TouchControllerWS.cpp
Fixes #125.
closed 2021-04-15 by TiVoHomeUser