ThingPulse/esp8266-weather-station-color #130: Moon Phases Appear Reversed

SillyHats, 2021-04-29:
In my location the the moon phase should be 88% illuminated as indicated by online calendars and the color weather clock itself. However the moon icon shows 88% dark with the left 12% illuminated. It looks like this could be fixed by switching between the upper and lowercase version of the font according the below font link. How can I switch moon font case?

[image]

https://www.dafont.com/moon-phases.font
```
  gfx.setFont(MoonPhases_Regular_36);
  gfx.setColor(MINI_WHITE);
  gfx.setTextAlignment(TEXT_ALIGN_CENTER);
  gfx.drawString(120, 275, String((char) (97 + (moonData.illumination * 26))));
```
https://github.com/ThingPulse/esp8266-weather-station-color/blob/c18d41038ca91ed18cd461029128a90514fffa56/esp8266-weather-station-color.ino#L442-L445

marcelstoer, 2021-05-02:
Ahh, these moon phases...You might get a better approximation if you replace 97 with 65.

Is that a 3D-printed case you have there?

closed 2021-05-02 by marcelstoer

SillyHats, 2021-05-02:
Thanks for the tip. I'll give it a try when I get a chance.

This is the case I printed 
https://www.thingiverse.com/thing:2172068

SillyHats, 2021-05-09:
Now that we have gone from waning gibbous to new moon the 3% illumination shows correctly with the original 97.
[image]

Switching to 97 to 65 reverses it to appear incorrectly 97% illuminated at this point in the cycle.

What is the source of the moon parameter data you referenced on the stack exchange discussion?

marcelstoer, 2021-05-09:
Turns I still had my playground code around; just pushed it: https://github.com/ThingPulse/esp8266-weather-station-color/compare/fix/moon. Not sure if it's any useful. I think I maneuvered myself into a dead-end with that `determineMoonIcon()` function. You can still opt to use the old method right there on this line https://github.com/ThingPulse/esp8266-weather-station-color/compare/fix/moon#diff-af80725fde10b4f3e4c3341c195be9f9e10cf47fbd2fb7d0c9cd2a55814874f8R442

> What is the source of the moon parameter data you referenced on the stack exchange discussion?

I may not fully understand that question, let me try.

That branch I talked about above has me use the full `SunMoonCalc` from our weather station library rather than the simplified `Astronomy` class. Hence, I have a ton of moon properties available to work with: https://github.com/ThingPulse/esp8266-weather-station/blob/40861dc2c1040319a310a7e4fe1c2cd331889693/src/SunMoonCalc.h#L48-L65

I think there's everything there we would need to call that remote service to fetch a nice moon image but I didn't look into which properties we would need to use.