ThingPulse/esp8266-weather-station-color #131: Forecasts not being grabbed

Brandoskey, 2021-06-19:
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
At boot forecasts are grabbed from open weather
### Actual behavior
weather station hangs at updating forecasts for a time then starts over at reconnecting to wifi and so on
### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.

```c
// add code here
```
### Weather Station Color version
Which branch are you on? If you know the Git revision then add it here as well.
Latest
### Hardware
Describe whether you run the code on the ThingPulse Weather Station Color kit or on some custom hardware.

I run on the color kit

If I comment out the following the weather stations finishes it's boot process but obviously has no weather data

`drawProgress(70, "Updating forecasts...");
  OpenWeatherMapForecast *forecastClient = new OpenWeatherMapForecast();
  forecastClient->setMetric(IS_METRIC);
  forecastClient->setLanguage(OPEN_WEATHER_MAP_LANGUAGE);
  uint8_t allowedHours[] = {6};
  forecastClient->setAllowedHours(allowedHours, sizeof(allowedHours));
  forecastClient->updateForecastsById(forecasts, OPEN_WEATHER_MAP_APP_ID, OPEN_WEATHER_MAP_LOCATION_ID, MAX_FORECASTS);
  delete forecastClient;
  forecastClient = nullptr;`

carlosimparato, 2022-02-18:
buenas. tengo un problema con la hora, en cada medio segundo cambia de 12H a 24H, he cambiado en settings.h la configuracion de (bool IS_STYLE_12HR = false;), de false a true y sigue igual una ayudita. gracias de ante mano

TiVoHomeUser, 2022-02-19:
**Brandoskey:** Did you find a solution? The only think I can think of is the memory is not setup correctly [Select flash mode and flash size](https://docs.thingpulse.com/guides/wifi-color-display-kit/#select-flash-mode-and-flash-size)

Brandoskey, 2022-02-19:
> **Brandoskey:** Did you find a solution? The only think I can think of is the memory is not setup correctly [Select flash mode and flash size](https://docs.thingpulse.com/guides/wifi-color-display-kit/#select-flash-mode-and-flash-size)

Very sorry for not updating this. I added a new unifi AP and the firmware had some sort of issue with the hardware I'm using for my weather station. An updated firmware fixed the issue as did moving to a location in the house where it would connect to another AP.

closed 2022-02-19 by Brandoskey