ThingPulse/esp8266-weather-station-color #132: NonTouchscreen

hayatibebek, 2021-06-22:
Hi, i use same display without touch fuction but my screen have a problem with clock style because always try to change 24/12 hour style how i can fix this.

Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

### Actual behavior

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.

```c
// add code here
```
### Weather Station Color version
Which branch are you on? If you know the Git revision then add it here as well.

### Hardware
Describe whether you run the code on the ThingPulse Weather Station Color kit or on some custom hardware.

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

marcelstoer, 2021-06-22:
Likely a dup of #84

closed 2021-06-22 by marcelstoer

hayatibebek, 2021-06-23:
thanks for help.

Marcel Stör ***@***.***>, 22 Haz 2021 Sal, 23:18 tarihinde
şunu yazdı: