ThingPulse/esp8266-weather-station-color #140: Stops updating weather conditions

MennoLabs, 2021-08-24:
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

Regularly updates weather conditions

### Actual behavior

Hangs at same conditions untill power cycled.

### Test code

### Weather Station Color version
Which branch are you on? If you know the Git revision then add it here as well.

latest
### Hardware
Describe whether you run the code on the ThingPulse Weather Station Color kit or on some custom hardware.

AZ-Delivery set:
https://www.az-delivery.de/en/products/az-touch-wandgehauseset-mit-2-8-zoll-touchscreen-fur-esp8266-und-esp32
ESP32-WROOM-32

8<------------------------ END BUG REPORT -------------------------------------

marcelstoer, 2021-08-24:
Please provide serial console output. Without it it's close to impossible to address this. The code itself is ok, runs fine here.

MennoLabs, 2021-08-24:
It just appears to stop updating. Right now showing 18.1 degrees, using the
API-call in my browser and Domoticz I get 17.6.
Terminal output:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Connecting to WiFi AIVD-surveillance/20396896797630894312
.Connected...Initializing touch screen...
Time difference for DST: 7200Getting url:
http://api.openweathermap.org/data/2.5/weather?id=2748076&appid=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx&units=metric&lang=nl
[HTTP] GET...
[HTTP] GET... code: 200
start document
Getting url:
http://api.openweathermap.org/data/2.5/forecast?id=2748076&appid=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx&units=metric&lang=nl
[HTTP] GET...
[HTTP] GET... code: 200
start document

DST Rules Updated:
DST Start: Sun Mar 28 03:00:00 2021
DST End:   Sun Oct 31 03:00:00 2021

Time difference for DST: 7200Getting url:
http://api.openweathermap.org/data/2.5/weather?id=2748076&appid=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx&units=metric&lang=nl
[HTTP] GET...
[HTTP] GET... code: 200
start document
Getting url:
http://api.openweathermap.org/data/2.5/forecast?id=2748076&appid=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx&units=metric&lang=nl
[HTTP] GET...
[HTTP] GET... code: 200
start document
Time difference for DST: 7200Getting url:
http://api.openweathermap.org/data/2.5/weather?id=2748076&appid=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx&units=metric&lang=nl
[HTTP] GET...
[HTTP] GET... code: 200
start document
Getting url:
http://api.openweathermap.org/data/2.5/forecast?id=2748076&appid=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx&units=metric&lang=nl
[HTTP] GET...
[HTTP] GET... code: 200
start document
Time difference for DST: 7200Getting url:
http://api.openweathermap.org/data/2.5/weather?id=2748076&appid=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx&units=metric&lang=nl
[HTTP] GET...
[HTTP] GET... code: 200
start document
Getting url:
http://api.openweathermap.org/data/2.5/forecast?id=2748076&appid=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx&units=metric&lang=nl
[HTTP] GET...
[HTTP] GET... code: 200
start document
Time difference for DST: 7200Getting url:
http://api.openweathermap.org/data/2.5/weather?id=2748076&appid=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx&units=metric&lang=nl
[HTTP] GET...
[HTTP] GET... code: 200
start document
Getting url:
http://api.openweathermap.org/data/2.5/forecast?id=2748076&appid=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx&units=metric&lang=nl
[HTTP] GET...
[HTTP] GET... code: 200
start document
Time difference for DST: 7200Getting url:
http://api.openweathermap.org/data/2.5/weather?id=2748076&appid=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx&units=metric&lang=nl
[HTTP] GET...
[HTTP] GET... code: 200
start document
Getting url:
http://api.openweathermap.org/data/2.5/forecast?id=2748076&appid=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx&units=metric&lang=nl
[HTTP] GET...
[HTTP] GET... code: 200
start document

Kind regards,
Menno

marcelstoer, 2021-08-24:
I can see you are not running an up-to-date version of the app. Please update as per https://docs.thingpulse.com/guides/wifi-color-display-kit/#prepare-arduino-ide-for-esp8266.

closed 2022-01-29 by marcelstoer