ThingPulse/esp8266-weather-station-color #145: Touchscreen looping back to calibration

tedtms, 2021-12-28:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
System should boot, prompt for calibration once, and then proceed with displaying the expected weather data.

### Actual behavior
When booting, the Featherwing touchscreen shows the connection attempt to my WiFi, which is successful. Then it shows a screen saying "Please calibrate touch screen by touch point" along with a white dot. Pressing the dot and the one that follows after works fine, but then an exception gets thrown and the system reboots, re-connecting to WiFi and again asking for calibration.

### Serial output (WiFi SSID and pass obfuscated)
```
Connecting to WiFi ssid/pass
......Connected...Initializing touch screen...
Mounting file system...
PIFFS opened: 
Calibration not available ###At this point, the TFT asks for calibration on two touchpoints. After pressing the second touchpoint, the following exception occurs:

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x40201332 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffdb0 end: 3fffffc0 offset: 0190
3fffff40:  3ffef6ac 3ffefd44 3ffef6ac 40201332  
3fffff50:  40216458 00000000 000003e8 00000000  
3fffff60:  00000000 00000000 00000000 4020cb74  
3fffff70:  00000000 00000000 3ffefd44 4020b18a  
3fffff80:  3ffef6ac 3ffef750 3ffefad4 40201dee  
3fffff90:  00000000 002c002f 00000000 feefeffe  
3fffffa0:  3fffdad0 00000000 3ffefc00 4020b204  
3fffffb0:  feefeffe feefeffe 3ffe8690 40100f51  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v000715a0
~ld
5
Connecting to WiFi ssid/pass
......Connected...Initializing touch screen...
Mounting file system...
PIFFS opened: 
Calibration not available
```
###Exception Decoder Result
```
Exception 0: Illegal instruction
PC: 0x40201332
EXCVADDR: 0x00000000

Decoding stack results
0x4020cb74: fs::FS::_defaultTimeCB() at /home/tedtms/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/FS.h line 249
0x4020b18a: __yield() at /home/tedtms/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/core_esp8266_main.cpp line 116
0x40201dee: setup() at /home/tedtms/Arduino/esp8266-weather-station-color/esp8266-weather-station-color.ino line 203
0x4020b204: loop_wrapper() at /home/tedtms/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/core_esp8266_main.cpp line 198
```

### Weather Station Color version
Master branch, installed via Arduino IDE library manager.

### Hardware
Feather HUZZAH with Featherwing TFT display. For testing, I'm using USB power instead of a LiPo battery.

8<------------------------ END BUG REPORT -------------------------------------

tedtms, 2021-12-28:
I followed the exception stack to lines 188 - 207 of esp8266-weather-station-color.ino, which contain the functions for touchscreen calibration. Since calibration of the touchscreen doesn't affect any of the weather-related features, I commented out those lines for testing purposes:

```

#ifdef HAVE_TOUCHPAD
  //SPIFFS.remove("/calibration.txt");
  //boolean isCalibrationAvailable = touchController.loadCalibration();
  //if (!isCalibrationAvailable) {
  //  Serial.println("Calibration not available");
  //  touchController.startCalibration(&calibration);
  //  while (!touchController.isCalibrationFinished()) {
  //    gfx.fillBuffer(0);
  //    gfx.setColor(MINI_YELLOW);
  //    gfx.setTextAlignment(TEXT_ALIGN_CENTER);
  //    gfx.drawString(120, 160, "Please calibrate\ntouch screen by\ntouch point");
  //    touchController.continueCalibration();
  //    gfx.commit();
  //    yield();
  //  }
  //  touchController.saveCalibration();
  //} else {
  //  Serial.println("Touchscreen calibrated");
  //}
#endif
```
Then I recompiled and uploaded. As expected, there was no exception thrown and the TFT showed (blank) weather data immediately after connecting to WiFi. Unfortunately I'm no developer and am unable to figure out _why_ the touchscreen calibration fails, but I expect it may be related to being (un)able to store calibration.txt after the user taps the two calibration dots.

marcelstoer, 2021-12-28:
If you're following the Adafruit tutorial you are effectively using _their_ forked code from https://github.com/adafruit/esp8266-weather-station-color/ -> closing

closed 2021-12-28 by marcelstoer

tedtms, 2021-12-28:
Yep, sorry, I just realized that and came here to close it. Looks like they have an open issue for this problem already. Thanks!