ThingPulse/esp8266-weather-station-color #148: LCD touch detection/calibration issue

blankana, 2022-01-28:
There seems to have been an issue with the calibration of my lcd and I'm unsure how to force a new calibration, the monitor shows only "Touch point detected at -1/-1" and "-1/0." where 90% of the screen is detected as -1/-1. The only thing I can do by touching the screen is switch between 12hr and 24hr time, naturally I'd like to use the other screens I'm seeing in videos demonstrating the weather station.

TiVoHomeUser, 2022-01-29:
You need to re-flash select "erase all flash contents" or to keep most of the saved data modify the setup() function after
 `drawProgress(100, "Formatting done"); `
insert 
 `if(ts.touched()){
  if(SPIFFS.exists("/calibration.txt")) SPIFFS.remove("/calibration.txt");
  while(ts.touched()){
    gfx.fillBuffer(MINI_BLACK); gfx.commit();
    delay(15); yield();
    gfx.fillBuffer(MINI_YELLOW); gfx.commit();
  }
}
 `
Then if you press the screen and hold while booting the calibration data will be deleted forcing the touch screen to re-calibrate.

blankana, 2022-01-29:
> You need to re-flash select "erase all flash contents" or to keep most of the saved data modify the setup() function after `drawProgress(100, "Formatting done"); ` insert `if(ts.touched()){ [...]

This did the trick, much obliged. I will note for anyone else doing this that you have to hold a touch on the screen _**after**_ the splash screen starts not before which causes a 'fatal error' manifesting as a locked black screen.

closed 2022-01-29 by blankana

marcelstoer, 2022-01-29:
@TiVoHomeUser thanks for sharing this interesting idea! What was the initial motivation for this "feature"? Was your device misbehaving in some way or did you just think it would be useful to have a non-invasive way to reset the touch calibration?

TiVoHomeUser, 2022-01-29:
It was a quick rough fix to this issue opened by blankana who could not change screens due to a bad touch screen calibration.
I do think it would be useful to be able to force a touch screen re-calibration without having to connect to a computer to wipe the memory.
 I need to test a better way would be to just set the isCalibrationAvailable boolean instead of deleting the file.

```
boolean isCalibrationAvailable = false
if(ts.touched()){
 /*  if(SPIFFS.exists("/calibration.txt")) SPIFFS.remove("/calibration.txt"); */
  while(ts.touched()){
    gfx.fillBuffer(MINI_BLACK); gfx.commit();
    delay(15); yield();
    gfx.fillBuffer(MINI_YELLOW); gfx.commit();
  }
} else {
  isCalibrationAvailable = touchController.loadCalibration();
}
/* boolean isCalibrationAvailable = touchController.loadCalibration();*/

```

Also working on 
Add a couple of seconds with a display screen instructing the press and hold and a confirmation release screen.
Move the connectWifi() to after the screen calibration not sure what that would break however.

marcelstoer, 2022-01-29:
> I do think it would be useful to be able to force a touch screen re-calibration without having to connect to a computer to wipe the memory.

👍 I fully agree and I like your ideas. Is this something you'll pursue and plan on raising a PR for? Would be awesome.

blankana, 2022-01-30:
> 👍 I fully agree and I like your ideas. Is this something you'll pursue and plan on raising a PR for? Would be awesome.

I'd have to agree, it was very easy for me to foul up the calibration because as far as I was aware I touched the dots exactly in the center with the stylus and despite a few hours and re-flashing multiple times I couldn't get a new calibration prompt to trigger. I would imagine a bad calibration could be an issue for others in the future.

TiVoHomeUser, 2022-01-30:
Created a new pull request "Allow user to force a screen re-calibration"

Unfamiliar with the new GitHub desktop I hope the request is correct.

TiVoHomeUser, 2022-01-30:
Added screen navigation
  Divided the screen into 4 sections.

A touch in section: 
- Top changes the time 12/24hr format.
- Left navigates back one page.
- Right navigates forward one page.
- Bottom jump to page 0.

PR #150 (2022-02-20): Update esp8266-weather-station-color.ino
 - Allow user to force a screen re-calibration
 - Break up the screen into 4 sections for navigation

Fixes #148.

- [x] This PR is compliant with the [other contributing guidelines](https://github.com/ThingPulse/esp8266-weather-station-color/blob/master/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] Should this code require changes to documentation I will contribute those to [https://github.com/ThingPulse/docs](https://github.com/ThingPulse/docs).

\<Description of and rationale behind this [... cut]