ThingPulse/esp8266-weather-station-color #154: Feature request: Probability of precipitation in %

Crover81, 2022-04-26:
8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature - Probability of precipitation in %

### Justification
I believe that visualizing this data is useful to understand how likely it is that it will rain / snow on that specific day
I noticed that by not displaying the probabilities the APIs give rain even at times when the probability is only 10% making the forecasts inaccurate

If it helps in the JSON file the probability data should be "pop" 0 = 0% 1 = 100%

8<------------------------ END FEATURE REQUEST --------------------------------

Thanks

marcelstoer, 2023-01-26:
That would have to be rendered separately as text I think. The icon to render is part of the OWM result set (`weather.icon`).

G6EJD, 2023-01-26:
They issue probability of precipitation as a per-unit value or if you like per-cent if their value is x100 that’s all I do in my projects then for example if 0.57 then display PoP=57%