ThingPulse/esp8266-weather-station-color #155: Once again - station does not pull data from Open weather map

MrSniperPhil, 2022-05-16:
So, as i've seen there have been issues with the station not pulling data from Open Weather Map, claiming to be fixed in various releases.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Plug station in, loggs into Wifi, pulls data from OWM
### Actual behavior
Plug station in, pulls correct time from server, apparently doesn't pull data from OWM, displays 0°C, no predictions etc.
Sunrise/set is at 01:00, all the prediction dates are set to Thursday 1:00
### Weather Station Color version
Latest github master

### Hardware
D1 Mini Pro ESP8266
2,8" SPI TFT Display

8<------------------------ END BUG REPORT -------------------------------------

I've tried a lot of different citys, from the ones already in the config files (Zurich and Sydney) to London, Nuremberg and Bayreuth (Bayreuth being the city i actually want to use).
I tried only inserting my Wifi information and API key, changing nothing else and it still does not work.

If i call the API manually on my Browser, it returns values as expected, so i assume that my API Key is valid.
Since the station calls the correct time, i assume that my Wifi also works fine.

No idea what i should try...

marcelstoer, 2022-05-16:
Posting the output from the serial monitor here would be helpful I guess.

MrSniperPhil, 2022-05-16:
[image]
[image]

I am assuming you're asking for this?

marcelstoer, 2022-05-16:
In the Arduino IDE: Tools -> Serial Monitor after you selected the correct port through Tools -> Port, then hit reset on the WeMOS module.

MrSniperPhil, 2022-05-16:
Complete gibberish:

0⸮~?⸮4⸮Ҷ⸮⸮OzI{⸮⸮8V⸮⸮wn⸮⸮`k⸮c⸮⸮⸮C⸮q⸮⸮~⸮⸮v⸮⸮P⸮㞅⸮g⸮f⸮⸮0⸮⸮⸮⸮)⸮⸮9⸮Ʀ
8⸮⸮⸮<⸮<⸮ӥ!Ȯ!⸮⸮⸮⸮⸮9F})⸮Ȯ9ƽ⸮⸮⸮⸮9F⸮1⸮
this is until it boots into the main screen

marcelstoer, 2022-05-16:
Doesn't look like the baud rate of the monitor was set to 115'200.

MrSniperPhil, 2022-05-16:
aaah, there we go:

Attempting to read application.properties file from SPIFFS.
Does not exist.
Effective properties now as follows:
	ssid: XXX
	password: XXX
	imezone: CET-1CEST,M3.5.0,M10.5.0/3
	OWM API key: 1215c3c608bd70c09e6c908eXXXXXX
	OWM location id: 2657896
	location name: Zurich
	metric: true
	12h style: false
15
Initializing touch screen...
Mounting file system...
SPIFFS opened: OK
Connecting to WiFi XXX/XXX......connected.
Connected, IP address: 192.168.0.37/255.255.255.0
Connected, MAC address: BC:FF:4D:4D:98:18
Configuring time for timezone CET-1CEST,M3.5.0,M10.5.0/3

Local time: Mon May 16 22:42:31 2022
UTC time:   Mon May 16 20:42:31 2022
[HTTP] Requesting resource at http://api.openweathermap.org:80/data/2.5/weather?id=2657896&appid=1215c3c608bd70c09e6c908eXXXXX&units=metric&lang=en
[HTTP] failed to connect to host
[HTTP] Requesting resource at http://api.openweathermap.org:80/data/2.5/forecast?id=2657896&appid=1215c3c608bd70c09e6c908eXXXXX&units=metric&lang=en
[HTTP] failed to connect to host
Free mem: 14272

so.. it can't connect to OWM, but when i use the URL in my browser it spits out stuff

MrSniperPhil, 2022-05-16:
Ok, i'm a dumbo...

My local DNS entry in my router still pointed to the old IP of my Pihole... now it works.

Sorry to waste your time. *sigh*

closed 2022-05-16 by MrSniperPhil