ThingPulse/esp8266-weather-station-color #156: Landscape mode

teargas666, 2022-06-13:
Hi. I have a problem with my station. I've already flash it and connect it with wires but my screen is rotated to landscape mode as it is on the image... [image]

teargas666, 2022-06-14:
I've used the wiring diagram from that video

https://youtu.be/eTIrP0__aGU

It's shown on 5:03

TiVoHomeUser, 2022-06-14:
Suggest you re-install [Mini Grafx](https://github.com/ThingPulse/minigrafx) by ThingPulse Library

teargas666, 2022-06-15:
I've reinstall it in Arduino on Windows and MacOS... no change...

closed 2022-06-15 by marcelstoer

teargas666, 2022-06-15:
Its just a photo from the phone, and it was not focused properly. I have a clear image, but just rotated... 

I've tried to change some code in the .ino file, and i can flip the screen horizontally and vertically... I still think its the software problem. Mayby I have to downgrade the Mini Grafx library...

TiVoHomeUser, 2022-06-15:
I had the previous version 1.2.5 of the Mini Grafx library installed I updated to 1.2.6 and still could not reproduce you're issue I programed with the "Erase all flash content" for a clean install.

It probably is a bad display you could try adding gfx.setRotation(0); after gfx.init(); in the setup() function.

 _gfx.init();_
  **gfx.setRotation(0);**
  _gfx.fillBuffer(MINI_BLACK);_

I rotated my screen with setRotation(3) and this is what Landscape mode would look like.
[image]
Notice the lack of the noise on the right. Another indication that the screen is damaged is that you only see the "Now release screen" message while the screen is being pressed removing you're finger the screen quickly changes to the "Please calibrate" message.

teargas666, 2022-06-16:
I cannot move to the calibration. I think I have to use the another screen. I use the setRotation(3) to receive the proper orientation

marcelstoer, 2022-06-16:
> I think I have to use the another screen

Apart from the [starter kit](https://thingpulse.com/product/esp8266-wifi-color-display-kit-2-4/) this project was designed for we are also selling [its display module](https://thingpulse.com/product/ili9341-2-4-spi-tft-lcd-touch-panel-breakout-board/) separately.

vicwhexagon, 2022-06-16:
I recommend the kit it eliminates the wires.
Where did you're screen come from?

vicwhexagon, 2022-06-24:
So we are on the same page which "screen" Did you get?
If you made some changes to the esp8266-weather-station-color code you could create a fork of the code then and post your changes there.

G6EJD, 2022-06-24:
Have you tried the Arduino example touch screen sketches to prove the screen and touch panel is working?

TiVoHomeUser, 2022-06-24:
- What changes have you made to the code?
 -  Where did you get the screen?

 - I suggest you download a fresh copy of this code
   - Change only the 2 WiFi credentials for WIFI_SSID and WIFI_PASS in settings.h so it can to connect to you're network
   - Compile and Upload the code using "Erase all flash content" to you're device. 
   - The display and touch screen should now function correctly without any weather data.
   - If that works  change the OpenWeatherMap settings for you're account to get the weather data for your chosen location.

teargas666, 2022-06-24:
There was just added the gfx.setRotation(0-3) - I've tried all of these possibilities. 

I've ordered one on the Shopee and second one on the Polish marketplace Allegro.

TiVoHomeUser, 2022-06-24:
There are a few on the site was it the 320 X 240 or 480 X 320 or some other?
This program uses the 320 X 240 display
Do you have a link?

teargas666, 2022-06-25:
This is following link https://shopee.pl/product/584766164/13553437087?smtt=0.756037666-1656136129.9

brianmiller1956, 2022-06-25:
Just an observation, this display as pictured does not have the touch controller fitted, It is not the correct display.

teargas666, 2022-06-25:
Yeah, there is not soldered touch controller but it's not a main issue...