ThingPulse/esp8266-weather-station-color #168: Wont conenct to Wifi

Zico36, 2023-12-08:
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Should connect to Wifi
### Actual behavior
Stuck in loading screen for wifi connection

### Weather Station Color version
Which branch are you on? If you know the Git revision then add it here as well.

### Hardware
Describe whether you run the code on the ThingPulse Weather Station Color kit or on some custom hardware.

8<------------------------ END BUG REPORT -------------------------------------

WGrav01, 2024-01-06:
Does your router support 2.4 ghz + has it on? The ESP8266 chip does not support 5ghz WiFi.

friedpa, 2024-01-11:
I have unfortunately overwriten the SW, after flashing the old Thingpulse SW again I also lost the Wifi. Something in the WiFi libary changed and the ESP8266 is not connecting anymore....

JohnyPeaN, 2024-04-08:
Got it working by setting WIFI_HOSTNAME to something shorter like "wstation". The default string is probably too long.

closed 2024-05-03 by marcelstoer