ThingPulse/esp8266-weather-station-color #171: ESP8266 Weatherstatiobn color locks up when updating from OpenWeatherMap

dtgrdn, 2024-04-28:
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Unit should update and show the eather

### Unit eventually locks up while trying to update the weather

```
### Weather Station Color version
Everything is current

### Hardware
I'm using the actual ESP8266 Weather Station Color that I bought from the ThingPulse website last month

8<------------------------ END BUG REPORT -------------------------------------

I've been pulling mt hair out trying to solve this issue. when my ESP8266 WeatherStation updates, it locks up. Either on "Updating Current Weather" or it will make it to "Updating forecasts", the progress bar just stops and it doesn't proceed back to the weather screen. I've gone through and read every issue and tried a few different suggestions for similiar issues, but it still does it.

I did notice that when it does update/work the last line in the Serial Monitor is "Free mem:", but when it fails/locks up the line isn't there.  

I've tried deleting everything and starting from scratch, but that hasn't worked either.

14:14:30.975 -> SDK:2.2.2-dev(38a443e)/Core:3.1.2=30102000/lwIP:STABLE-2_1_3_RELEASE/glue:1.2-65-g06164fb/BearSSL:b024386
14:14:31.022 -> Attempting to read application.properties file from SPIFFS.
14:14:31.076 -> Does not exist.
14:14:31.076 -> Effective properties now as follows:
14:14:31.076 ->     ssid:
14:14:31.076 ->     password:
14:14:31.076 ->     imezone: PST8PDT,M3.2.0,M11.1.0
14:14:31.076 ->     OWM API key:
14:14:31.076 ->     OWM location id: 5746545
14:14:31.076 ->     location name: Portland
14:14:31.076 ->     metric: false
14:14:31.076 ->     12h style: true
14:14:31.076 -> 15
14:14:31.277 -> Initializing touch screen...
14:14:31.277 -> Mounting file system...
14:14:34.397 -> SPIFFS opened: OK
14:14:34.397 -> Connecting to WiFi xxxxx/xxxxxxxxfpm close 3
14:14:34.397 -> mode : sta(50:02:91:e9:fd:35)
14:14:34.397 -> add if0
14:14:35.046 -> .....scandone
14:14:38.261 -> state: 0 -> 2 (b0)
14:14:38.261 -> state: 2 -> 3 (0)
14:14:38.314 -> .state: 3 -> 5 (10)
14:14:38.314 -> add 0
14:14:38.314 -> aid 18
14:14:38.314 -> cnt
14:14:38.862 -> .
14:14:39.317 -> connected with xxxxx, channel 8
14:14:39.317 -> dhcp client start...
14:14:39.464 -> ....ip:192.168.1.39,mask:255.255.255.0,gw:192.168.1.1
14:14:41.722 -> .connected.
14:14:41.806 -> Connected, IP address: 192.168.1.39/255.255.255.0
14:14:41.806 -> Connected, MAC address: 50:02:91:E9:FD:35
14:14:41.806 -> Configuring time for timezone PST8PDT,M3.2.0,M11.1.0
14:14:41.847 -> .
14:14:42.225 -> Local time: Sun Apr 28 14:15:33 2024
14:14:42.225 -> UTC time:   Sun Apr 28 21:15:33 2024
14:14:42.310 -> [HTTP] Requesting resource at http://api.openweathermap.org:80/data/2.5/weather?id=5746545&appid=b31a21ccd7d38e7c25fe130dea43477b&units=imperial&lang=en
14:14:42.356 -> [HTTP] connected, now GETting data
14:14:42.356 -> [String] 'GET /data/ ... hermap.org': Reallocating large String(133 -> 156 bytes)
14:14:42.356 -> start document
14:14:42.441 -> [HTTP] Requesting resource at http://api.openweathermap.org:80/data/2.5/forecast?id=5746545&appid=b31a21ccd7d38e7c25fe130dea43477b&units=imperial&lang=en
14:14:42.495 -> [HTTP] connected, now GETting data
14:14:42.495 -> [String] 'GET /data/ ... hermap.org': Reallocating large String(134 -> 157 bytes)
14:14:42.495 -> start document
14:14:42.842 -> Free mem: 12912
14:14:48.363 -> pm open,type:2 0
14:24:31.150 -> [HTTP] Requesting resource at http://api.openweathermap.org:80/data/2.5/weather?id=5746545&appid=b31a21ccd7d38e7c25fe130dea43477b&units=imperial&lang=en
14:24:31.197 -> [HTTP] connected, now GETting data
14:24:31.197 -> [String] 'GET /data/ ... hermap.org': Reallocating large String(133 -> 156 bytes)
14:24:31.197 -> start document
14:24:31.297 -> [HTTP] Requesting resource at http://api.openweathermap.org:80/data/2.5/forecast?id=5746545&appid=b31a21ccd7d38e7c25fe130dea43477b&units=imperial&lang=en
14:24:31.350 -> [HTTP] connected, now GETting data
14:24:31.350 -> [String] 'GET /data/ ... hermap.org': Reallocating large String(134 -> 157 bytes)
14:24:31.350 -> start document

dtgrdn, 2024-04-30:
more info, locked up when updating conditions:
stopped on "Updating conitions"

[HTTP] Requesting resource at [http://api.openweathermap.org:80/data/2.5/weather?id=5746545&appid=b31a21ccd7d38e7c25fe130dea43477b&units=imperial&lang=en](http://api.openweathermap.org/data/2.5/weather?id=5746545&appid=b31a21ccd7d38e7c25fe130dea43477b&units=imperial&lang=en)
[hostByName] request IP for: api.openweathermap.org
[hostByName] Host: api.openweathermap.org IP: 38.89.70.82
:ref 1
[HTTP] connected, now GETting data
[String] 'GET /data/ ... hermap.org': Reallocating large String(133 -> 156 bytes)
:wr 156 0
:wrc 156 156 0
:ack 156
:rn 536
start document
:c0 1, 536

TiVoHomeUser, 2024-04-30:
I tried with your resource for Portland it works for me.
 What are you using to compile and upload?
 Which chip are you selecting? LOLIN (WEMOS) D1 & R2 Mini is what I have set.

TiVoHomeUser, 2024-04-30:
Note I don't get the reallocating message check your strings.

> [String] 'GET /data/ ... hermap.org': **Reallocating large String(133 -> 156 bytes)**

Local time: Tue Apr 30 08:23:27 2024
UTC time:   Tue Apr 30 15:23:27 2024
[HTTP] Requesting resource at http://api.openweathermap.org:80/data/2.5/weather?id=5746545&appid=b31a21ccd7d38e7c25fe130dea43477b&units=imperial&lang=en
[HTTP] connected, now GETting data
start document
[HTTP] Requesting resource at http://api.openweathermap.org:80/data/2.5/forecast?id=5746545&appid=b31a21ccd7d38e7c25fe130dea43477b&units=imperial&lang=en
**[HTTP] connected, now GETting data**
start document
Free mem: 9784

dtgrdn, 2024-04-30:
> I tried with your resource for Portland it works for me. What are you using to compile and upload? Which chip are you selecting? LOLIN (WEMOS) D1 & R2 Mini is what I have set.

I'm selecting LOLIN (WEMOS) D1 & R2 Mini and configuring it per the ThingPulse setup document. I've got the current Arduino IDE 2.3.2 and current libraries.

dtgrdn, 2024-04-30:
I messed around with this today. Removed everything and started from scratch, again. I started with ESP8266 WeatherStation ver 2.0.0 library. It compiled and I didnt' received the "Reallocating large string". I then updated the library to 2.2.0 and still didn't receive the "Reallocating large string" message. I let it sit and it worked/ updated every 10 minutes for 1 hour. At the 1 hour update it locked up at "Updating current conditions". I'm at a total loss here and have put in way too much time trying to get something so simple to work.

dtgrdn, 2024-05-02:
Daniel suggested checking my wifi on my post on the ThingPulse support website. I found that QoS was turned on. After turning it off my ThingPulse Weather station connects and runs flawlessly. I'm closing this as "human error" :)

Thanks for the support!
Dave

closed 2024-05-02 by dtgrdn