ThingPulse/esp8266-weather-station-color #172: Stuck on yellow calibration screen

knight-yyz, 2024-07-13:
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Plug USB port into microcontroller and the chip should boot to the weather station
### Actual behavior
After being unplugged for a few days and plugged back in the Weather station rushes through all the screens and stops at blue calibration screen for a few seconds.  When I touch the screen the wemos led does not light up.  After a few seconds the yellow screen says release.  The yellow screen stays.  Now when I touch the screen the blue led lights up on the wemos.  
### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.

```c
// add code here
```
### Weather Station Color version
Which branch are you on? If you know the Git revision then add it here as well.
2018 thingpulse
### Hardware
Describe whether you run the code on the ThingPulse Weather Station Color kit or on some custom hardware.
ali express tft
8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

disaster999, 2024-09-03:
Having a similar issue. The weather stations boots up and tells me to press and hold the screen for calibration which I did. Then a yellow screen pops up telling me to release, which I did, but the yellow screen stays. I did this numbers of time, reset the device number of times and that yellow screen remains. Repeated touching of the screen does not light up the wemos LED at all anywhere during this "calibration" phase. Please help.

TiVoHomeUser, 2024-09-03:
Does it go through the two calibration touches?
The next step is the WiFi connection.

disaster999, 2024-09-04:
> Does it go through the two calibration touches? The next step is the WiFi connection.

No, It just freezes on the yellow calibration window. No amount of presses or touches anywhere on the screen does anything.

marcelstoer, 2024-09-04:
@disaster999 Here's the commit Ron refers to c52d098056ba3446f69f856e895aa8fe935c3d8a

closed 2024-09-04 by marcelstoer

disaster999, 2024-09-05:
> @disaster999 Here's the commit Ron refers to [c52d098](https://github.com/ThingPulse/esp8266-weather-station-color/commit/c52d098056ba3446f69f856e895aa8fe935c3d8a)

this did not fix my issue as the screen is still stuck at the yellow screen calibration window. At this point im thinking my screen is defective more than anything. Is there a way to check if the touch screen I got works or not?